compas-dev/compas_slicer #102: linear_velocity for print_organization

cheibasethz, 2021-01-19:
# Linear_velocity update

Dear Ioanna and Joris, 
I would like to ask for your help and expertise to update the linear_velocity with a new velocity_type named "various". The idea is to print designated areas of the geometry faster, while others remain slower. 

## Details

To give an example: there is a .stl geometry in the link below, which has certain areas as double surfaces, therefore duplicate points. What I would like the code to do is to determine which points are duplicate based on proximity, and go faster in that area. Unfortunately, there are some errors that I do not know how to fix. Can you give me some suggestions?

## Code
The files in the wetransfer link: https://we.tl/t-rUpZaa0xbl

The "various" new velocity_type added in the linear_velocity file looks something like this:

    a = []
    b = []
    printpoints = []
    def calculate_linear_velocity_based_on_points(printpoint, mesh, proximity_range, speed_range):
    """
    Calculates the linear velocity for duplicate points, so that the velocity is faster on proximity to a duplicate point.

    Parameters
    ----------
    printpoint: :class: 'compas_slicer.geometry.PrintPoint'
    mesh: :class: 'compas.datastructures.Mesh'
    proximity_range: tupple
    speed_range: tupple
    """
    # instead of printing two surfaces with min_distance close points, 
    # the function should print only one set of points and enhance the velocity

    # determine the proximity points  
    network = Network()
    printpoints.append(printpoint)
    tree = KDTree(printpoints)

    for point in printpoints:
        network.add_node(x=point[0], y=point[1], z=point[2])

    for node in network.nodes():
        point = network.node_coordinates(node)
        for nbr in tree.nearest_neighbors(point, 1, distance_sort=True):
            if nbr[2] < 1e-6:
                #printpoints real
                a.append(node)
                continue
            if not network.has_edge(node, nbr[1], directed=False):
                # printpoint duplicate
                b.append(node)
    proximity_distance = distance_point_point(a, b)
    # remove the duplicate points
    mesh_delete_duplicate_vertices(mesh)

    # remap values
    min_speed, max_speed = speed_range
    min_distance, max_distance = proximity_range
    old_range = max_distance - min_distance
    new_range = max_speed - min_speed

    if proximity_distance <= min_distance:
        return max_speed
    elif min_distance < proximity_distance <= max_distance:
        velocity = (((proximity_distance - min_distance) * new_range) / old_range) + min_speed
        return new_range - (velocity - new_range)
    else:  # proximity_distance > max_distance:
        return min_speed

Obviously, this code can be entirely changed if you have a better idea of how to update the function :)

ioanna21, 2021-01-20:
Hello, thanks for your message! 
Looking at your code I have first some comments for minor things/errord, and secondly a question.

On the `01_slicing_beta`, line 15 the default compas.slicer `set_linear_velocity `is imported, I assume you want to use your modified function in  linear_velocity_beta.py, right? The replace line 15 with `from linear_velocity_beta import set_linear_velocity`

In line 183 you call the function `mesh_delete_duplicate_vertices `that is not declared or imported anywhere, so this throws me an error. Did you create this function? Could you include it in the file?

Line 170 the error ‘printpoint is not subscritable’ comes up. You should replace `network.add_node(x=point [0], y=point [1], z=point [2])` with `network.add_node(x=point.pt[0], y=point.pt[1], z=point.pt[2])`

Finally line 182 the compas function `distance_point_point` is called with arguments the two lists a,b. To the best of my knowledge, this takes two points as arguments, and that is why it throws an error. Which distance are you trying to calculate there?

Questions 
Looking at your function, the part that assigns linear velocity to the PrintPoints looks good, it should work well assuming that your remapping is correct. It is the part that finds the proximity to existing points on the mesh that is the tricky part. You need to find which points have another segment in great proximity, without considering their immediate neighbors. So you create a `Network `and a `KDTree `and somehow use those for the proximity check. This part is not so clear to me. Could you explain what is the logic behind these, i.e. how you intend this to work?

ioanna21, 2021-01-20:
Thinking about the proximity check and finding where the double surfaces are, in general I think this is easier to do during the generation of geometry, instead of trying to reconstruct the information in a post-processing step. If you have access to the geometry generation step, maybe it's easier to intervene there either by changing those surfaces, or for example by adding related attributes to the mesh faces/vertices that could then be passed to the printpoints.

closed 2021-07-22 by ioannaMitropoulou