compas-dev/compas_slicer #109: When sorting into vertical layers, a safety_printpoint is always added even if it's not always necessary

joburger, 2021-02-01:
A safety printpoint should be added if there are multiple Vertical Layers on the same height, however if there is only one VerticalLayer (so one branch) it is not necessary to add a safety point, see image:

[image]

ioanna21, 2021-02-05:
Aha I see this, it's coming from example 3 right? 
This is happening because we have the `max_paths_per_vertical_layer `value quite low, so it splits the vertical_layers in smaller vertical layers. When a vertical layer finishes, then it adds an interruption without checking where the next vertical layer starts. So I wonder how we would want to fix this. I.e. when vertical layers are split, doesn't it make sense that there's an interruption in between? I.e. in this case wouldn't it make more sense to have one continuous vertical layer? But then it would have to 'violate' the `max_paths_per_vertical_layer ` value. Or do we want to check between vertical layers and only add an interruption if they are in a distance higher than a threshold? The second solution seems a bit prone to mistakes because we would have to hardcode a threshold value. I am not sure what is the right solution

joburger, 2021-02-05:
Yea exactly, I think maybe one possible approach could be that we use the `min_max_z_height` to check if there is more than one `VerticalLayer`, and if there is only one `VerticalLayer`, there does not have to be an interruption added? Because I think the desired behaviour is that there is an interuption when there are multiple VerticalLayers at the same height, but no interuption when there is only one. What do you think?