compas-dev/compas_slicer #137: error during preprocessing [curved slicing]

inesariza, 2021-06-26:
Hi! Just trying out the curved slicer for the first time, so I may be missing important info on how to generate the targets:

**Describe the bug**
When running the simple curved interpolation slicing, with the vase example or custom inputs, I get the following error:

```
INFO - Loaded json: c:/Users/arizai/Dropbox (Personal)/phd/waam_cairo\data\boundaryLOW.json
INFO - Loaded json: c:/Users/arizai/Dropbox (Personal)/phd/waam_cairo\data\boundaryHIGH.json
INFO - Creating target with attribute : boundary=1
INFO - union_method : min, union_params =  []
INFO - Compound target with 'boundary'=1. Number of connected_components : 1
C:\Users\arizai\Anaconda3\envs\waam_cairo\lib\site-packages\compas_slicer\pre_processing\preprocessing_utils\geodesics.py:41: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the 
ndarray.
  f = np.array(f)
Bad Typechar 'O'
Traceback (most recent call last):
  File "c:/Users/arizai/Dropbox (Personal)/phd/waam_cairo/vase_slice_non-planar.py", line 45, in <module>
    preprocessor.create_compound_targets()
  File "C:\Users\arizai\Anaconda3\envs\waam_cairo\lib\site-packages\compas_slicer\pre_processing\interpolation_slicing_preprocessor.py", line 56, in create_compound_targets
    geodesics_method=geodesics_method)
  File "C:\Users\arizai\Anaconda3\envs\waam_cairo\lib\site-packages\compas_slicer\pre_processing\preprocessing_utils\compound_target.py", line 82, in __init__
    self.compute_geodesic_distances()
  File "C:\Users\arizai\Anaconda3\envs\waam_cairo\lib\site-packages\compas_slicer\pre_processing\preprocessing_utils\compound_target.py", line 113, in compute_geodesic_distances
    self.clustered_vkeys]
  File "C:\Users\arizai\Anaconda3\envs\waam_cairo\lib\site-packages\compas_slicer\pre_processing\preprocessing_utils\compound_target.py", line 112, in <listcomp>
    distances_lists = [get_igl_EXACT_geodesic_distances(self.mesh, vstarts) for vstarts in
  File "C:\Users\arizai\Anaconda3\envs\waam_cairo\lib\site-packages\compas_slicer\pre_processing\preprocessing_utils\geodesics.py", line 44, in get_igl_EXACT_geodesic_distances
    distances = igl.exact_geodesic(v, f, vstart, vertices_target)
ValueError: Invalid scalar type (object, Col Major) for argument 'f'. Expected one of ['int32', 'int64', 'int128'].

```

**To Reproduce**
Steps to reproduce the behavior:
1. Context [e.g. ST3, Rhino, Blender, ...]: Targets created with curved_slicing_master.gh (attached)
2. Sample script: https://compas.dev/compas_slicer/latest/examples/02_curved_slicing_simple.html#
3. Sample data (attached) 
[curved_slicer_files.zip](https://github.com/compas-dev/compas_slicer/files/6720396/curved_slicer_files.zip)

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Win10
 - Python version [e.g. 2.7] 3.7
 - Python package manager [e.g. macports, pip, conda] conda

Let me know if you need more info to reproduce. Thank you!

ioannaMitropoulou, 2021-06-30:
Hello, thanks for sharing this!
The error comes up when it tries to calculate the geodesic distances using the libigl function, but it's a bit hard to tell from this why exactly this is happening.
Could you share with me the files you are using for this slicing so that I can look into it in more detail? 
Thanks!

brgcode, 2021-06-30:
@inesariza can you post the contents of `f`?

ioannaMitropoulou, 2021-06-30:
Looking into the files, the problem seems to be the topology of the mesh; it has quad faces, while compas_slicer only works for triangular faces. As a result, f throws an error because has rows of 3 and 4 elements. 
[image]

I will add a check for the mesh input, so that it catches quad meshes early in the process and throws a nicer error that explains what the problem is, to avoid future confusion.

I remeshed and triangulated the input mesh to a nicer topology, I attach the new files at the end of this message.

[image]

If you use this mesh input, then the slicing works.

[image]

Give this a try, and let us know if more issues come up.

[curved_slicer_files.zip](https://github.com/compas-dev/compas_slicer/files/6740064/curved_slicer_files.zip)

inesariza, 2021-07-02:
thank you very much for your replies. all working perfectly! 👍

closed 2021-07-02 by inesariza