compas-dev/compas_slicer #153: matplotlib dependencies prevents from use in Rhino

jf---, 2021-12-14:
the following import statement:
```python
from compas_slicer.post_processing import generate_brim
```

seems to rely on matplotlib

**Expected behavior**
matplotlib is tricky / impossible to install to on `IronPython`
hence this dependency renders `compas_slicer` incompatible with Rhino
moreover, since matplotlib is a pretty hefty package ( can depend on `Qt5` in turn ) it takes a whopping ~10s to load

**Desktop (please complete the following information):**
 - OS: osx 10.10, Rhino 6
 - IronPython
 - compas 1.21.1
 - compas_slicer ( HEAD )

**Screenshots**
[image]

[image]

jf---, 2021-12-14:
- factored out `matplotlib`, loading is still _slow_
- btw on compas python this all pans, out, this is `IronPython` / Rhino specific
- hmmm this is suspicious:

https://github.com/compas-dev/compas_slicer/blob/58dedfbebd123258506174453d5f73d18d745819/src/compas_slicer/__init__.py#L63-L71

- yep, commenting the above out reduces the time to import from 11 -> 1s
- that said, the anti-pattern is applied broader:

https://github.com/compas-dev/compas_slicer/blob/58dedfbebd123258506174453d5f73d18d745819/src/compas_slicer/post_processing/__init__.py#L45-L65

joburger, 2021-12-16:
Hi Jelle, so actually compas slicer is, at the moment, not designed to work directly within Rhino. Reason for this is to avoid all of the constraints of IronPython. So the envisioned workflow is as follows:

- Export STL/OBJ from your favorite software
- Run the Python slicing file (outside of Rhino)
- Json files with slicing result are generated
- Visualize json files in Grasshopper (or using compas viewers)
- Repeat slicing if adjustments are needed

This is somehow described here (https://compas.dev/compas_slicer/latest/tutorials/02_grasshopper_visualization.html), but I realize it's probably not clear enough so will add some documentation to describe this workflow better. 

Also, we are not considering making compas slicer available for current versions of Rhino, since CPython will be introduced in Rhino8, we would probably wait until R8 is out and then direct some efforts to integrate compas slicer into a Rhino workflow.

Cheers,
Joris

closed 2025-12-16 by jf---