compas-dev/compas_slicer #156: Exception when running script ex2_curved_slicing_advanced.py

yvanblanchard, 2022-02-22:
Hello,

when runnning the ex2_curved_slicing_advanced.py script, I get errors at the topological sorting step. I did not changed the original script and input files.

Thank you

[image]

ioannaMitropoulou, 2022-02-23:
Hello! Could you make sure you have pulled the latest version of compas_slicer? 
I just ran this example, it seems to work. The resulting paths visualized using the grasshopper component 'csLoadSlicer' are the following:
[image]
If after updating to the latest compas_slicer you still run into this problem, could you zip the entire example folder and send it to me to see if I can reproduce it? 

Thanks,
Ioanna

yvanblanchard, 2022-02-23:
Hello,
I pulled the latest state of code, and also the new example python script (the latest one uses 'implify_paths_rdp_igl' with IGL).
But I finally have the same issue (assertion error when doing topological sorting).
Here is my example folder , containing the resulting files, zipped.
[data_advanced_example-ybd.zip](https://github.com/compas-dev/compas_slicer/files/8127541/data_advanced_example-ybd.zip)

yvanblanchard, 2022-02-24:
Hello, Waiting for an explanation (I hope!) using my provided data set, could it be possible to get the json outpoints files for this shape please ?
thank you very much for your help.

ioannaMitropoulou, 2022-02-25:
Hello,
I haven't managed to reproduce the error. When I ran the example on the files you sent me, I get the expected results. I send you back the folder with the data you sent me, the python script I used for the slicing (just a copy of the example2), all the resulting jsons, and the grasshopper file for visualizing the results.

[image]

[example-ybd.zip](https://github.com/compas-dev/compas_slicer/files/8139266/example-ybd.zip)

closed 2024-03-12 by ioannaMitropoulou