compas-dev/compas_slicer #170: Installation of Compas Slicer on windows

OjasMahajan, 2024-09-20:
Hello, I need help with installing the slicer on my system.
I use a Windows system and work with Rhino 7. 
I have constantly got this error when I try to follow your installation steps

**WAAM was the environment that I created.**

PS C:\Users\oxido> & C:/Users/oxido/anaconda3/envs/WAAM/python.exe "x:/WAAM/Documentation/Compas slicer/compas_slicer - Installation_files/compas_slicer-master/compas_slicer-master/examples/2_curved_slicing/ex2_curved_slicing.py"
Traceback (most recent call last):
  File "x:/WAAM/Documentation/Compas slicer/compas_slicer - Installation_files/compas_slicer-master/compas_slicer-master/examples/2_curved_slicing/ex2_curved_slicing.py", line 2, in <module>
    from compas.datastructures import Mesh
ModuleNotFoundError: No module named 'compas'
PS C:\Users\oxido>

Can you help me with this error?
Thanks..

chenkasirer, 2024-09-23:
Hi @OjasMahajan!

Could you please provide some more context regarding your issue?
What command did you use to install COMPAS slicer and what are you trying to execute and where?
Are you successfully activating the environment before running the script?

The only thing that pops out from the error you added is that you're using PowerShell. If you used `conda` to install I would highly recommend using the Anaconda Prompt, as it will for sure provide you everything you need to activate and use the correct environment.

OjasMahajan, 2024-09-23:
Hello @chenkasirer,
I tried installing everything again. And for now, it has started working. 
But now I have issues with the mesh adjusting the geodesic distance for this piece. can you help with this.

chenkasirer, 2024-09-23:
I unfortunately never used the slicer so have little to no idea.
@ioannaMitropoulou, perhaps you could help?

OjasMahajan, 2024-09-23:
For me cutting the mesh into 3 parts has solved the problem
[image]

ioannaMitropoulou, 2024-09-23:
Hello, 
Thanks for your messages! Did I understand well that the problem has been resolved? Let me know if there is still something I can help you with. 
Best,
Ioanna

closed 2024-11-05 by ioannaMitropoulou