compas-dev/compas_slicer #171: ImportError: cannot import name 'mesh_bounding_box' from 'compas.datastructures'

Arom122, 2024-11-06:
Hi, i can not run the tutorial after installation using following comand (python -m compas_slicer)
Traceback (most recent call last):
  File "D:\compas_slicer-master\examples\1_planar_slicing_simple\example_1_planar_slicing_simple.py", line 5, in <module>
    import compas_slicer.utilities as utils
  File "C:\Users\codefab\anaconda3\envs\my-project\lib\site-packages\compas_slicer\__init__.py", line 64, in <module>
    from .slicers import *  # noqa: F401 E402 F403
  File "C:\Users\codefab\anaconda3\envs\my-project\lib\site-packages\compas_slicer\slicers\__init__.py", line 35, in <module>
    from .base_slicer import *  # noqa: F401 E402 F403
  File "C:\Users\codefab\anaconda3\envs\my-project\lib\site-packages\compas_slicer\slicers\base_slicer.py", line 12, in <module>
    from compas.datastructures import mesh_bounding_box
ImportError: cannot import name 'mesh_bounding_box' from 'compas.datastructures' (C:\Users\codefab\anaconda3\envs\my-project\lib\site-packages\compas\datastructures\__init__.py)

**Desktop (please complete the following information):**
 - OS: Win10 anaconda 3

gigan5712, 2024-11-08:
same here..
Also Win10 and Anaconda 3

Arom122, 2024-11-21:
Thank you! I will try compass 1.16.0. @gigan5712